urob/zmk-config #1: Bluetooth not working on Advantage 360 Pro

deze333, 2023-02-09:
The firmware seems to work fine with the exception for the Bluetooth connectivity, it doesn't seem to be implemented?

urob, 2023-02-09:
It should be automatically working when `CONFIG_WIRELESS` is set to `1` as in here:
https://github.com/urob/zmk-config/blob/016b1f84dbe5e757fefbcbee358ccdf072071a91/config/adv360pro.keymap#L23

If you want to link your keymap I can take a look and see if I see what's going wrong.

deze333, 2023-02-10:
I have figured it out. The original Kinesis firmware has keyboard bluetooth profiles that remember laptop connection. This repo doesn't support those bluetooth profiles and the keyboard may occasionally get stuck. The workaround is to restore original Kinesis  firmware and use `Mod` + `GUI` to make the keyboard forget its bluetooth profile, and connect to laptop again. After that this repo firmware works fine.

Is it possible to restore the `Mod` layer functionality? Especially combinations that mount the halves.

urob, 2023-02-10:
I am not sure I follow. The bluetooth drivers used in this repo and the Kinesis one are exactly the same. The only difference is that I use different keybindings to switch between the different bluetooth profiles. But apart from that there should be no difference in functionality.

Specifically, the keys to switch between bluetooth profiles and to clear them are all located on the `SYS` layer: https://github.com/urob/zmk-config/blob/016b1f84dbe5e757fefbcbee358ccdf072071a91/config/base.keymap#L251-L261

If you prefer different mappings, you may also find my [adv360-demo-config](https://github.com/urob/adv360-demo-config) useful, which should be easier to understand and to use as a starting point for further customization.

deze333, 2023-02-10:
Got that. Confusion was due to the lack of LED profile indication – Kinesis changes LED color according to profile selected. Is that possible to make the LED work?

urob, 2023-02-10:
> Got that. Confusion was due to the lack of LED profile indication – Kinesis changes LED color according to profile selected. Is that possible to make the LED work?

Yes, if you swap the west.yml file in this repo with this one from the led-branch in my 360-demo repo it should work:

https://github.com/urob/adv360-demo-config/blob/led/config/west.yml

The readme file in the demo-repo contains further details. Note in particular that the led functionality is super experimental and there isn't much support by the zmk community for it.

deze333, 2023-02-11:
It worked, thank you! Just swapping one line for `revision: adv360-led` did the trick. Much nicer now.

One more question if you don't mind. If I wanted to run build locally (to avoid endless pushes and errors), is there a description of how to do that? The Kinesis process is neat, compiling locally is a breeze.

urob, 2023-02-11:
> It worked, thank you! Just swapping one line for `revision: adv360-led` did the trick. Much nicer now.
>
>
>
> One more question if you don't mind. If I wanted to run build locally (to avoid endless [...]

You can use the build script in this repo. See the readme for details:

https://github.com/urob/zmk-config/blob/main/scripts/readme.md

Alternatively the official zmk instructions are here:

https://zmk.dev/docs/development/setup

brGuirra, 2023-05-30:
Hello, @urob! First things first, I want to thank you for the amazing work!

I'm trying to enable the led feature but I'm stuck with problems during the build process. I thought I've messed up something while making a few tweaks with the keymap, but even when I get a fresh fork of your repo and just change the revision to `adv360-led`, I still face errors like this during the build process:

[image]

Do you have any idea of why this happens? This is my [fork](https://github.com/brGuirra/zmk-config) for reference.

urob, 2023-05-30:
The led branch is a bit outdated as Kinesis hasn't yet updated their drivers for the new zephyr 3.2. 

I don't remember fully but I think the 3.0 legacy version of this repo should work together with the adv360-led branch of the zmk repo.

brGuirra, 2023-06-01:
> The led branch is a bit outdated as Kinesis hasn't yet updated their drivers for the new zephyr 3.2.
> 
> I don't remember fully but I think the 3.0 legacy version of this repo should work together with the adv360-led branch of the zmk repo.
> 
> Edit: or just try replacing `sys-reset` with `reset` in your keymap, which is the change from 3.0 to 3.2 that cause the error you are seeing.

Replacing `sys-reset` with `reset` solve the problem, thanks!

closed 2023-06-26 by urob

tjallemaal, 2023-07-02:
I'd also like to turn on LED. I included the `revision: adv360-led` line into `build.yaml` but get an error, may I ask what I'm doing wrong?

`include:
  - board: adv360pro_left
  - board: adv360pro_right
revision: adv360-led`

urob, 2023-07-03:
> I'd also like to turn on LED. I included the `revision: adv360-led` line into `build.yaml` but get an error, may I ask what I'm doing wrong?
> 
> 
> 
> ```
> 
> include:
> 
>   - board: adv360pro_left
> 
>   - board: adv360pro_right
> 
> revision: adv360-led
> 
> ```

Wrong file, you'll have to edit to west.yaml:

https://github.com/urob/zmk-config/issues/1#issuecomment-1425874480

tjallemaal, 2023-07-06:
I'm trying and it doesn't work :/

I forked the main branch. Now when I take `west.yaml` from the `adv36-demo-config` and replace my `build.yaml` with it, this error occurs:

[image]

urob, 2023-07-06:
> I'm trying and it doesn't work :/
> 
> 
> 
> I forked the main branch. Now when I take `west.yaml` from the `adv36-demo-config` and replace my `build.yaml` with it, this error occurs:
> 
> 
> 
> [image]
> 
> 
You need to replace the west.yml file not the build.yaml. The contents of the file that you are replacing should be almost identical to the one I linked above.

tjallemaal, 2023-07-08:
rob, I'm really really sorry. I did as you said but I still get an error. I'ma give up now.
[image]
[image]

urob, 2023-07-08:
> rob, I'm really really sorry. I did as you said but I still get an error. I'ma give up now.

Ok, if you ever wanna try again, check my comment on this issue from May 30 as well as to you in the other issue. The answer is right there.