urob/zmk-config #2: Mouse layer doesn't work over bluetooth

yanshay, 2023-03-21:
I'm using the Kinesis config as base but with urob/zmk adv360-led branch (using west.yaml).
I'm using Mac.

The mouse works but only over USB and not over bluetooth.  
I saw in the zmk repo questions about that and everyone said that repairing fix this, but not in my case, so I wonder maybe something is special about the mouse support in this branch and/or due to me mixing Kinesis config with the adv-360 branch and that cause some issues?

Any ideas?

infused-kim, 2023-09-12:
It seems like that is an issue exclusive to the adv360 branch, because it adds some capslock LED characteristics to the BT profile.

You can find more info and a potential solution in this zmk discord discussion:
Quick correction: The mouse PR on urob's fork is working fien with macOS Ventura 13.5.

My issue was caused, because I had another PR that added BT characteristics on top of this PR in `app/src/hog.c`.

If someone else encounters similar issues, you can check this discussion on the zmk discord:
https://discord.com/channels/719497620560543766/845285481888743434/1151183774235762818

If someone else experiences mouse keys not working, check if you have any other PRs that modify `app/src/hog.c` and adjust it according to the info in the discord chat.

urob, 2025-02-19:
Closing this as mouse emulation is now in upstream. If anyone is  still having issues you may want to open a bug report there.

closed 2025-02-19 by urob