urob/zmk-config #3: Do you know of anyone who forked this for the glove 80?

danielo515, 2023-03-30:
Hello. Thank you for sharing your configs. It seems to be very good thought. I'm particularly attracted by the quick home row mods and the combos. I'm coming from qmk and I heavily rely on those.
I am in the process of moving to ZMK for my new globe80, but there is a lot to digest and configure. I was wondering if you know someone that is using your configs for the glove 80, so I can just steal their config 😁

pujvz, 2023-03-31:
I am interested in this as well. Please let me know if anyone forks this or if support is added for the glove80.

urob, 2023-03-31:
Hey, thanks for your interest!

I am not aware of anyone using this config for the glove80 so far, but I also haven't really kept track of all the forks.

In either case, since both the combos and homerow-mods are configured using [key position helpers](https://github.com/urob/zmk-nodefree-config#key-position-helpers), it would be straightforward to just add the combos and homerow mods to any other layout. The main step would be to create a keyposition definition for the glove80. Here are some examples for other keyboards: https://github.com/urob/zmk-nodefree-config/tree/main/keypos_def.
(As an aside, Stephen was at some point interested in incorporating some of the helper macros from the nodefree-repo in the factory default firmware for the glove80. So it might be worth checking with him whether he has a keypos definition flying around).

For the rest of the configuration, I would assume that you probably would make some changes to make better use of the many more keys on the glove80. But just in case you want to start out with my keymap, you can take a look how I incorporate it into  [my configuration for the Kinesis 360](https://github.com/urob/zmk-config/blob/main/config/adv360pro.keymap), which in terms of layout is pretty similar to the glove80. The main step would be defining the `X_` macros to pad the extra keys of the layout. But again, that would probably not be the most sensible keymap for the glove80...

pujvz, 2023-04-01:
Thanks for the well documented config urob :smiley: . 

@danielo515 I cloned glove80 zmk and cherry picked PR #1387 and #1423. I tested it with a single build and it is working well with `global-quick-tap-ms = <100>`. Try at your own risk if you want. Assuming you already have moergo-sc/glove80-zmk-config, you'll need to edit .github/workflows/build.yml and change the repository to mine.

ie find this section in build.yml and edit the repository like so
```
      - uses: actions/checkout@v3
        with:
          repository: pujvz/zmk
          ref: main
          path: src
```