urob/zmk-config #4: Did you implement less than and greater than combos?

rhhub, 2023-04-02:
In the layout picture, the RM1-RM2 and RM2-RM3 combos are shown with parenthesis and alligator symbols. I don't think this is implemented.

rhhub, 2023-04-02:
Oh, nevermind they are implemented under your nav layer. I thought they were accessed through shift.

closed 2023-04-02 by rhhub

urob, 2023-04-03:
For `[ {` and `] }` you can use either shift or Nav. I'm still deciding which I like more, and whether to add shift for the other ones as well.

rhhub, 2023-04-03:
I noticed that. It's what initially caused my confusion. I think either works. I'm probably going to give shift a shot. 

On a side not I think your macros are awesome and definitely make combos more manageable. I even have a gaming laying that I implemented 0-9 on the verticals of a choc ferris sweep. They seem to work okay given my minimal testing in gameplay.