urob/zmk-config #5: star and fslh combo is not working

chris-ryu, 2023-04-09:
"star"와 "flslh" 콤보가 작동 하지 않습니다.

당신의 리포지터리를 바탕으로 Ferris Sweep 34key용 펌웨어를 작성 중입니다.

 combos.dtsi등의 파일은 그대로 복사해서 사용하고  base.keymap을 다음과 같이 수정 했습니다.

아무래도 "RM2" 키가 위 아래로 작동하는데 관련이 있는 것 같습니다만 정확한 이유를 찾지 못하겠습니다.

도움을 요청 드립니다.

#include <behaviors.dtsi>
#include <dt-bindings/zmk/keys.h>
#if CONFIG_WIRELESS
    #include <dt-bindings/zmk/bt.h>
    #define _BT_SEL_KEYS_ &bt BT_SEL 0 &bt BT_SEL 1 &bt BT_SEL 2 &bt BT_SEL 3 &bt BT_CLR
#else
    #define _BT_SEL_KEYS_ &trans &trans &trans &trans &trans
#endif

#include "helper.h"

#define DEF 0
#define NAV 1
#define FN 2
#define NUM 3
#define SYS 4
#define UC 5
#define MOUSE 6

#include "combos.dtsi"  // requires layer aliases
#include "mouse.dtsi"
#include "extra_keys.h"

#define XXX &none
#define ___ &trans

/* Global settings */

#define QUICK_TAP_MS 175

&sk {  // sticky-key config
    release-after-ms = <600>;  // release after 0.6s
    quick-release;             // no double capitalization when rolling keys
};

&sl {  // sticky-layer config
    ignore-modifiers;          // allow chording sticky mods & layers
};

&lt {  // layer-tap config
    flavor = "balanced";
    tapping-term-ms = <200>;
    quick-tap-ms = <QUICK_TAP_MS>;
};

/* Homerow mods */

#define KEYS_L LT0 LT1 LT2 LT3 LT4 LM0 LM1 LM2 LM3 LM4 LB0 LB1 LB2 LB3 LB4  // left hand
#define KEYS_R RT0 RT1 RT2 RT3 RT4 RM0 RM1 RM2 RM3 RM4 RB0 RB1 RB2 RB3 RB4  // right hand
#define THUMBS LH1 LH0 RH0 RH1                                      // thumbs

#define MAKE_HRM(NAME, HOLD, TAP, TRIGGER_POS) \
    ZMK_BEHAVIOR(NAME, hold_tap, \
        flavor = "balanced"; \
        tapping-term-ms = <280>; \
        quick-tap-ms = <QUICK_TAP_MS>; \
        global-quick-tap-ms = <150>; \
        bindings = <HOLD>, <TAP>; \
        hold-trigger-key-positions = <TRIGGER_POS>; \
        hold-trigger-on-release; \
[... 227 lines cut ...]
ZMK_LAYER(sys,
//╭──────╮ ╭─────────────┬─────────────┬─────────────┬─────────────┬─────────────╮ ╭──────╮ ╭─────────────┬─────────────┬─────────────┬─────────────┬─────────────╮ ╭──────╮
    X_LT                                _BT_SEL_KEYS_                                X_MT     ___           ___           ___           ___           ___             X_RT
//├──────┤ ├─────────────┼─────────────┼─────────────┼─────────────┼─────────────┤ ├──────┤ ├─────────────┼─────────────┼─────────────┼─────────────┼─────────────┤ ├──────┤
    X_LM     ___           ___           ___           ___           &bootloader     X_MM     &bootloader   ___           ___           ___           ___             X_RM
//├──────┤ ├─────────────┼─────────────┼─────────────┼─────────────┼─────────────┤ ├──────┤ ├─────────────┼─────────────┼─────────────┼─────────────┼─────────────┤ ├──────┤
    X_LB     ___           ___           ___           ___           &sys_reset      X_MB     &sys_reset    ___           ___           ___           ___             X_RB
//├──────┤ ╰─────────────┼─────────────┴─────────────┼─────────────┼─────────────┤ ├──────┤ ├─────────────┼─────────────┼─────────────┴───────────────────────────╯ ├──────┤
    X_LH                                               ___           ___             X_MH     ___           ___                                                       X_RH
//╰──────╯                                           ╰─────────────┴─────────────╯ ╰──────╯ ╰─────────────┴─────────────╯                                           ╰──────╯
)

ZMK_LAYER(mouse,
//╭──────╮ ╭─────────────┬─────────────┬─────────────┬─────────────┬─────────────╮ ╭──────╮ ╭─────────────┬─────────────┬─────────────┬─────────────┬─────────────╮ ╭──────╮
    X_LT     ___           ___           ___           ___           ___             X_MT     ___           U_WH_D        U_MS_U         U_WH_U       ___             X_RT
//├──────┤ ├─────────────┼─────────────┼─────────────┼─────────────┼─────────────┤ ├──────┤ ├─────────────┼─────────────┼─────────────┼─────────────┼─────────────┤ ├──────┤
    X_LM     ___           ___           ___           ___           ___             X_MM     ___           U_MS_L        U_MS_D         U_MS_R       ___             X_RM
//├──────┤ ├─────────────┼─────────────┼─────────────┼─────────────┼─────────────┤ ├──────┤ ├─────────────┼─────────────┼─────────────┼─────────────┼─────────────┤ ├──────┤
    X_LB     ___           ___           ___           ___           ___             X_MB     ___           ___           ___            ___          ___             X_RB
//├──────┤ ╰─────────────┼─────────────┴─────────────┼─────────────┼─────────────┤ ├──────┤ ├─────────────┼─────────────┼─────────────┴───────────────────────────╯ ├──────┤
    X_LH                                               ___           ___             X_MH     U_BTN2        U_BTN1                                                    X_RH
//╰──────╯                                           ╰─────────────┴─────────────╯ ╰──────╯ ╰─────────────┴─────────────╯                                           ╰──────╯
)

/* vim: set ft=c tw=174: */

urob, 2023-04-09:
Can you send a link to your repo?

chris-ryu, 2023-04-09:
> Can you send a link to your repo?

https://github.com/chris-ryu/zmk-config

urob, 2023-04-09:
You'll have to increase the max-combos-per-key setting to at least 6 for the latest combo setup to work. Like this:
https://github.com/urob/zmk-config/blob/b03dda12311ab2e858918176c376cf1213f957dc/config/corneish_zen.conf#L14-L16

chris-ryu, 2023-04-10:
👍🏼  Thank you so much.
I followed your advice and it fixed the problem.

I really like the new backspace key position and the left thumb position of the return key.  
I really appreciate your sharing.

closed 2023-04-10 by chris-ryu