urob/zmk-config #7: Question: How do you type vertical combos?

lkhphuc, 2023-04-25:
Sorry this might sound stupid, as I have never tried a split keyboard yet, still waiting for mine to arrive.

I find the idea very interesting, but I can't quite get how can you type the vertical combos comfortably. For horizontal combo, it is obvious that you use 2 fingers. But do you also use 2 fingers for 2 keys on the same column, or do you, idk, swipe them, or just press on the press of the 2 keys to activate the at once?

rhhub, 2023-04-25:
I have a [Ferris sweep Sweep2.x](https://github.com/davidphilipbarr/Sweep) with choc spacing, MBK Choc Keycaps and 20g linear switches. The edges of the keys are almost completely flush. I hit my verticals with one finger against a total actuation force of 40g (The same as a red). I actually find it easier than the horizontals while possibly a little slower; there is less target than a normal key.

urob, 2023-04-25:
>  I hit my verticals with one finger against a total actuation force of 40g.

Pretty much this. It definitely feels better with lighter switches and choc spacing, but even without I find it works well enough for the less frequent symbols.

closed 2023-04-26 by lkhphuc