urob/zmk-config #8: Nav.

RATIU5, 2023-04-28:


urob, 2023-04-28:
A single tap of `Nav` + `L` should send a single `backspace`. Holding  `Nav` + `L` should send `Ctrl` + `backspace`. If a single tap keeps sending backspace after you released the key, it sounds like you are experiencing some form of stuck keys. 

Which board are you using? I hadn't have an issue with stuck keys for a long while since some stuff has been fixed upstream. The one exception is my Adv360, which I have been struggling with lately and haven't yet debugged (I am waiting for Kinesis to bring the drivers up to date with the latest ZMK updates before further looking into it).

RATIU5, 2023-04-28:
Unfortunately, yes, it is the Adv360 giving me these issues. I've been trying it for a while and it seems to vary when it does it and when it doesn't. What version are they current ZMK Adv360 drivers at?

urob, 2023-04-28:
I am using https://github.com/zmkfirmware/zmk/pull/1454 at the core. You could try using the drivers from [KinesisCorporation/Adv360-Pro-ZMK](https://github.com/KinesisCorporation/Adv360-Pro-ZMK) instead, which would require the following steps:
* copy driver files into zmk-config from https://github.com/KinesisCorporation/Adv360-Pro-ZMK/tree/V2.0/config/boards/arm/adv360
* revert ZMK to 3.0 and patch with https://github.com/ReFil/zmk/tree/adv360-z3
* rearrange layout to match the matrix definition of those drivers
* replace `sys_reset` with `reset`

RATIU5, 2023-04-29:
Don't you have a layout that matches the Adv360 matrix? In your zmk-nodefree-config repo?
I attempted to follow your steps but I'm not experienced enough with C/ZMK to get a working build.

I tried to use the same build process and files as Kinesis has and made sure everything was reverted to 3.0. No luck on my end [https://github.com/RATIU5/zmk-config/tree/advantage](https://github.com/RATIU5/zmk-config/tree/advantage). I will possibly try to come against this later. In the meantime, I may revert layouts since some keys don't work until I spam them.

urob, 2025-02-19:
Closing this as I no longer maintain a custom Kinesis branch since any issues I had were resolved upstream. There have also been multiple major upgrades to both upstream ZMK and the Kinesis forks in the meantime, making this discussion quite dated. Hopefully your issues have been resolved as well. If not, you may want to open an bug report upstream.

closed 2025-02-19 by urob