urob/zmk-config #10: Support for nice_nano/kyria keyboard

mateuszaaa, 2023-05-04:
Hi, im quite impressed by your work, wannted to give it a shot but i cannot force it to compile for nice nano v2, is it compatible?

i followed guide for [scripts/readme.md](https://github.com/mateuszaaa/zmk-config-urob) and just migrated my config but it didnt work, so i decided to just use yours config but use `nice_nano_v2` board, but it fails the same way...

[log.txt](https://github.com/urob/zmk-config/files/11395526/log.txt)

am i missing something ? ;0

urob, 2023-05-04:
Using the script is just a small bonus feature to simplify compiling locally. Unfortunately the script doesn't currently support shields. There's an easy fix but for now you can also just compile your usual way; e.g., through GitHub Actions. Did that work for you?

mateuszaaa, 2023-05-04:
i tired to follow the guide [on reddit](https://www.reddit.com/r/ErgoMechKeyboards/comments/11gejh3/lpt_try_urobs_zmk_timeless_homerow_mods_combos/) and ended up doing with [such changes](https://github.com/mateuszaaa/zmk-config/commit/9ba8b83af1f5c9541e932a95cc8bda1c564429b0) but i think it does not work great as when im typing fast using home rows shift is not activated (i need to keep button pressed for some time to activate mod)

urob, 2023-05-04:
Oh wow, I hadn't seen the reddit guide. Thanks @infused-kim for putting that together!

A couple of things:

* You want to adjust the `#include "../../zmk-nodefree-config/keypos_def/keypos_36keys.h"` at the beginning of the file to match the Kyria, otherwise the positional-hold-taps will be off and you have to hold the HRMs past the tapping-term. I don't think I currently have one for the Kyria, but if you look at the examples here it should be easy to make one: https://github.com/urob/zmk-nodefree-config/tree/main/keypos_def
* HRM-shift can be a bit tricky to use for capitalization. That's why I recommend in the readme to use a dedicated thumb-shift for capitalization. But this is just me, and once you fix the positional hold-taps it may work for you. Alternatively, you could lower or disable the global-quick-tap just for shift as suggested by @infused-kim. You'll have to try out whether this works better for you.
* I am not sure if you have seen it, but my readme also has a small troubleshooting section suggesting a few things to try out to address certain issues with the HRM configuration.

urob, 2023-05-04:
Actually this should work for the Kyria, right? https://github.com/urob/zmk-nodefree-config/blob/main/keypos_def/keypos_58keys.h

If so, you can just replace the include at the top of the file with this:
```
#include "../../zmk-nodefree-config/keypos_def/keypos_58keys.h"
```

mateuszaaa, 2023-05-05:
just figured it out :0, wow man whats an amazin job it finnally works fine!!!

created PR with keypos for kyria, consider merging it if you think it has any value ;)

https://github.com/urob/zmk-nodefree-config/pull/16

mateuszaaa, 2023-05-05:
also leaving my migration config for anyone interested ;)
https://github.com/mateuszaaa/zmk-config/commit/3afc4ce24d0abbb92f09e9cb7b6f919a3a27a3eb

urob, 2023-05-05:
Great! Glad it worked and you like it. Thanks for putting up the PR with the Kyria keypos file. I left a small comment there. Once addressed it would be great to merge it.