urob/zmk-config #12: Selecting window from alt+tab (swapper)

ziombo, 2023-06-06:
How are you selecting a window with the current swapper implementation? Once I go into alt+tab mode, it requires me to send another key input to select a window, but the keypress affects selected window. 

For example if I'm switching between windows and press enter while selecting notepad, a new line will be inserted.

urob, 2023-06-06:
That doesn't sound right. Just releasing the Nav-key should focus the currently selected window. E.g., `hold Nav` -> `tap swapper once` -> `release Nav` will focus the previous window.

ziombo, 2023-06-06:
Ah, this might explain it as I have swapper assigned to a key on my main layer so there's no holding Nav

closed 2023-06-07 by ziombo