urob/zmk-config #13: Ctrl + Shift + ; and Ctrl + ; are unavailable

rhhub, 2023-06-21:
This is more of a commentary then a necessity. I'm sure it's your personal build after all.

Google sheets, and I think excel, have the following shortcuts:
1. To insert the current date: Ctrl + ; 
2. To insert the current time: Ctrl + Shift + ; (Ctrl + :)

Attempting these will produce < >. 

If no longer in use, because < > are accessible with combos, I suggest they be removed.

My working snippet.
```
// tap: comma | shift + tap: semicolon | ctrl + shift + tap: <
ZMK_BEHAVIOR(comma_morph, mod_morph,
    mods = <(MOD_LSFT|MOD_RSFT)>;
    bindings = <&kp COMMA>, <&kp SEMICOLON>; // updated to semicolon on this line
)
// Not needed greater than is on a combo
// ZMK_BEHAVIOR(comma_inner_morph, mod_morph,
//     bindings = <&kp SEMICOLON>, <&kp LESS_THAN>;
//     mods = <(MOD_LCTL|MOD_RCTL)>;
// )

// tap: dot | shift + tap: colon | ctrl + shift + tap: >
ZMK_BEHAVIOR(dot_morph, mod_morph,
    bindings = <&kp DOT>, <&kp COLON>; // updated to colon on this line
    mods = <(MOD_LSFT|MOD_RSFT)>;
)
// Not needed greater than is on a combo
// ZMK_BEHAVIOR(dot_inner_morph, mod_morph,
//     bindings = <&kp COLON>, <&kp GREATER_THAN>;
//     mods = <(MOD_LCTL|MOD_RCTL)>;
// )
```

urob, 2023-06-22:
True. I never needed those shortcuts, so I didn't think much about. I probably should just remove them since I fully got used to the new combos by now. If one would like to make the shortcuts work with the old mod-morphs, one could drop the `RCTRL` from the inner mod-morph like so:
```
ZMK_BEHAVIOR(comma_inner_morph, mod_morph,
    bindings = <&kp SEMICOLON>, <&kp LESS_THAN>;
    mods = <(MOD_LCTL)>;
)
```
and replace `&hmr LCTRL E` with `&hmr RCTRL E` on the keymap. This way one could trigger the shortcuts using the mods on the right side of the keyboard and still use the left-hand mods to morph `;` and `:` into `<` and `>`. But if you are fine using the new combos for `<` and `>`, then removing the inner morphs altogether like you did is definitely the cleanest way.

chris-ryu, 2024-02-02:
@urob I've noticed that in the latest version, it seems that you were made modifications related to a certain feature by implementing the SIMPLE_MORPH macro. I'm currently looking into how I might remove or bypass this particular modification for my project's needs. Could you please provide some guidance or suggest the best approach to achieve this?