urob/zmk-config #14: 360 build

tjallemaal, 2023-06-25:
I'm a total noob, I have no clue how to build the firmware. 
Watched some tutorials so I have an idea, 
but how do I build firmware from those multiple files that are compatible with not only the 360 pro?

urob, 2023-06-25:
You only need to build the 360 file and can ignore the rest. Some of the other files will get called automatically when building but as long as they are in the repo you don't need to do anything. The easiest way to build is to use Github Actions and download the compiled firmware from the Github Actions tab.

If you are only building for the 360, you could also start with this repo https://github.com/urob/adv360-demo-config, which is less complex.

tjallemaal, 2023-06-25:
Thanks alot!

tjallemaal, 2023-06-25:
I'm sorry to ask this, but how exactly do I build the `adv360pro.keymap` file only?
Do I have to move it into a separate directory? because with the main branch in `actions` I cannot select that file specifically

urob, 2023-06-25:
You can configure which boards are build with GitHub Actions in this file: https://github.com/urob/zmk-config/blob/main/build.yaml

Or you could just not care that it's building more boards and only download the firmware you need.

tjallemaal, 2023-06-25:
thanks so much really. I must have done sth wrong beforehand, as when I built, I didnt get the firmware at all.
Now I have it and it is just a matter of picking the right file after the download, lol.

Don't know quite exactly what I missed but it's working now

cheers

closed 2023-06-25 by tjallemaal