urob/zmk-config #15: no stroke recognition / flashing problem?

tjallemaal, 2023-06-25:
I think I successfully flashed the firmware or so i thought.
My keyboard is being recognised but upon typing I dont see any output, alas. I'm wondering what could have gone wrong here

urob, 2023-06-25:
It sounds like it isn't paired correctly. If you had it paired before and your previous build didn't have mouse-keys enabled, then you will need to re-set the pairing: 1. Forget bluetooth device on windows, 2. clear bluetooth connection on the 360, 3. re-pair

(This is necessary the first time you are enabling mouse keys, because the OS needs to change the driver from keyboard to keyboard + mouse)

tjallemaal, 2023-06-25:
unfortunately, still not working.
I'm on MacOS Catalina fwiw

maybe for some reason, my OS doesnt get to do` the mouse + keyboard` driver thing?

tjallemaal, 2023-06-26:
I just realised: it works when plugged in! it's a bluetooth thing.

closed 2023-06-26 by tjallemaal

urob, 2023-06-28:
Another thing you could try if you haven't gotten Bluetooth to work yet:

https://zmk.dev/blog/2023/04/06/zephyr-3-2#known-issues

Afaik the official Kinesis firmware is still running on zephyr 3.0. That's why this steps may gut be necessary

tjallemaal, 2023-06-28:
Oh! you just made me realise: when trying to BT_CLR I pressed the default 360-macro being `mod + gui`.

So when thinking of reset, I actually did not.

...
But I don't actually know how to access the `sys` layer yet.
How would that work?

urob, 2023-06-29:
I have leds disabled in my main branch. For the layout see the image in the readme.

tjallemaal, 2023-06-30:
I still don't get it :/ 

There is this line of code `#define _BT_SEL_KEYS_ &bt BT_SEL 0 &bt BT_SEL 1 &bt BT_SEL 2 &bt BT_SEL 3 &bt BT_CLR
`

I know to Fn+Num and then the BT key. But what do I do to cycle to `BT_CLR`?
Sorry, I really am kinda retarded

[image]

urob, 2023-06-30:
> I still don't get it :/ 
> 
> 
> 
> There is this line of code `#define _BT_SEL_KEYS_ &bt BT_SEL 0 &bt BT_SEL 1 &bt BT_SEL 2 &bt BT_SEL 3 &bt BT_CLR
> 
> `
> 
> 
> 
> I know to Fn+Num and then the BT key. But what do I do to cycle to `BT_CLR`?
> 
> Sorry, I really am kinda retarded
> 
> 
> 
> [image]
> 
> 
> 
> 

_BT_SEL_KEYS_ expands to the definition above. So BT_CLR is on the inner index position in that row.

tjallemaal, 2023-07-09:
> _BT_SEL_KEYS_ expands to the definition above. So BT_CLR is on the inner index position in that row.

What exactly do you mean by that? 
Could you possibly give an example where for example BT_2 is being cleared? I pressed the BT_SEL key and the key right to that simultaneously but I failed once again ...