urob/zmk-config #18: ZMK_UNICODE_PAIR prints the unicode value

raydevs, 2023-07-24:
Sorry for opening an issue for this matter, I really appreciate your work, have everything working but `ñ`.
I'm using your submodule in my configuration and merged with another fork you zmk fork in my repo.
I'm trying to print the Spanish special character ñ, but got the unicode printed instead.

`ZMK_UNICODE_PAIR(n_tilde,      N0, N0,  F, N1,    N0, N0,  D, N1)`

Using it like this:
`&n_tilde`

It's weird the fact that other special characters defined in your submodule are working fine.
Any ideas?

urob, 2023-07-28:
Hm, looks right to me. Maybe link your config

raydevs, 2023-07-28:
I forgot about putting the link: https://github.com/raydevs/zmk-config

urob, 2023-07-28:
Still no idea, but here are a few leads:

What exactly happens when you press `&n_tilde`? What happens when you press `LSHFT` along with it? 

Does connecting via usb make a difference?

Can you try again the Greek symbols on the Unicode layer? Do they all work on your current computer?

Just to be sure: you are trying to trigger `n_tilde` in the right-most column of the bottom row as defined on line 251 of base.keymap, right? That is, the definitions in corne_keymap and corne.keymap never make it to the actual keymap.

raydevs, 2023-07-29:
I have cleaned up the repo letting just the corne.keymap. The behavior is the same over bluetooth as via usb.
If I press `LSHFT` along with it: **u00d1**. 
Pressing only `&n_tilde`: **u00f1**.

Now the greek symbos don't even work. I remember they worked the first time I flash your config.
Any way, I changed the west file to use your **ZMK** fork and no luck, prints UC codes.

urob, 2023-07-29:
Given that Greek symbols don't work, maybe worth checking that your OS set up matches the Unicode set up:

https://github.com/urob/zmk-nodefree-config#dependencies-for-unicode

urob, 2023-07-29:
Right now, your repo doesn't specify a Unicode method, so it implicitly assumes that you are on windows (which is the default) and that you have WinCompose configured

raydevs, 2023-07-29:
I intend to use the keyboard on both Windows and Mac. Right now I'm testing on windows, so I'm keeping the default host.
I did a [test ](https://github.com/raydevs/zmk-config/blob/main/config/test.keymap) based in your example and just added `ZMK_UNICODE_PAIR(n_tilde,      N0, N0,  F, N1,    N0, N0,  D, N1)`, surprinsingly it works. So I will start again from the example adding the rest of behaviors.

Thanks!

closed 2023-07-29 by raydevs

raydevs, 2023-08-02:
I discovered why it worked with the example.keymap. Using spanish keyboard in windows makes "SEMI" prints "ñ". That solves my problem.