urob/zmk-config #19: Suggestion for German Umlauts on Nav instead of the rudendant "<" and ">"

cari66ean, 2023-07-31:
As per issue #13, having < and > on the combos works just fine. Therefore removing them from the Nav Layer allows all the German Umlauts to be placed there identically as they are currently only on the "Greek" layer. That's because ä and ö would replace < and > and the other 2 keys for ß and ü are coincidentally already free on the Nav layer as well.

This allows the German Umlauts to be typed much more easily, while keeping the same placement they currently inhibit on the less efficient to activate Greekt layer and while only removing 2 redundant keys that cause issues with shortcuts anyways.

urob, 2023-07-31:
Sounds reasonable. I don't use them that much to justify changing it on my keyboards. But for people that use them more often this sounds like a good change, which should be pretty easily to implement in their fork.

cari66ean, 2023-07-31:
Yeah it's indeed a very "German" variation and not useful for anyone else.