urob/zmk-config #20: COMBO_TERM_SLOW potentially causes false activations of mouse and Greek layers

cari66ean, 2023-07-31:
Very often when typing the word "analysis" I seem to have activated some weird shortcuts. It happened so often that it got really annoying. 

After "analysing" (sheesh) what's going on, I realized it's due to the "LY" bigram that is used to also activate the Greek layer as a combo. Now I don't think I noticed the same on the mouse layer, that is activated with the other hand on the W and P keys, but it's safe to say the "LY" bigram is not at all rare and thus causing false activations if pressed very quickly in succession. 

When looking at the combos.dtsi, I noticed combos of horizontally adjacent keys use COMBO_TERM_FAST, while combos with vertically adjacent keys use COMBO_TERM_SLOW. That absolutely makes sense. 

However, I noticed that COMBO_TERM_SLOW is also used for the combos for the mouse layer, the greek, the shifted greek layer and for cutting. These are combos that consist of keys that are on the same horizontal row as well but either not directly adjacent or require 3 keys. 

Now, I would argue that it makes no difference in speed if I need to press a combo with the index finger + middle finger or if I have to press it with the index finger + ring finger. It can be done equally as fast and thus COMBO_TERM_SLOW is not necessarily needed here. Furthermore, since they are so qluick to press at the same time OR in succession when typing, it allows quickly typing bigrams such as "LY" to be mistakingly registered as a combo. Which is why I would highly suggest COMBO_TERM_FAST for these.

Now trigrams such as "LUY" in case of the Shifted Greek layer or even more so "XCD" in case of the cut combo are probably quite rare, so mistakingly activating those combos when typing should be less of an issue. But I feel like pressing 3 keys with the index, middle and ring finger at the same time is also quick and easy enough to do, that one could also comfortably use COMBO_TERM_FAST here as well.

cari66ean, 2023-07-31:
For perspective I have also increased my Combo global_quick_tap_ms from 100 to 150 because I experienced false activations of other combos on the home rows (might even increase it further to 175, although 200 was too much). So that's already generally reducing false activations when typing, yet the "LY" bigram was still really bugging me. Now I'm using it with COMBO_TERM_FAST (that's further reduced from 20 to 18) and it seems perfect. Even the 3-key combos for Shifted Greek or Cut works perfectly fine, without causing false activations any longer.

urob, 2023-07-31:
This is a good call and makes totally sense. I also started noticing occasional false activation of the unicode combo since switching to ultra-light switches (25gf activation force). So I'll probably do something similar in my repo once I have played around with it a bit more.

cari66ean, 2023-07-31:
Yeah give it a try. Or maybe add a third COMBO_TERM option with a value somewhere in between. But from what I'm seeing it works perfectly fine with COMBO_TERM_FAST.