urob/zmk-config #25: Define extra keys for different layers

mind33, 2023-11-19:
Hello,
I am building a sofle keyboard. How do I define the extra keys on the keyboard for different layers?

urob, 2023-11-20:
Unfortunately, the current implementation isn't flexible enough for that. While it would certainly be possible to define layer-specific extra keys, it seems to me that at that point the additional complexity outweighs the gains of keeping all different keyboards in one repo and one may be better off splitting them up.

mind33, 2023-11-21:
Manage to do it by renaming the variables for each layers. e.g. I rename X_LT to X_LT0, X_LT1, X_LT2 and so for the different layers. It's not very elegant but it does the job.
 I have encountered issue when adding the sensor bindings. Initially I did not add the bindings for the sensor and the rotary encoders button (when pressed) worked. However once I added the bindings for the encoders (which work fine), nothing happen when I press the rotary encoder buttons.

urob, 2023-11-22:
Not sure about the encoder bindings. In case you use the `ZMK_LAYER` macro to set them up, the 3rd argument expands to `sensor-bindings` under the hood:

```cpp
#define ZMK_LAYER_3_ARGS(name, layout, sensors) \
    / { \
        keymap { \
            compatible = "zmk,keymap"; \
            layer_ ## name { \
                label = ZMK_HELPER_STRINGIFY(name); \
                bindings = <layout>; \
                sensor-bindings = <sensors>; \
            }; \
        }; \
    };
```

I don't use sensors myself but from what I understand this is the correct way to set them up. I'd check with the ZMK discord server to see if anyone knows something about the interaction between sensor-bindings and rotary buttons.