urob/zmk-config #50: Build script compat with shields

JJ-Atkinson, 2024-04-06:
First off, if this isn't a goal of the project, please let me know and just close the issue :).

So, after some effort I got podman installed and figured out how the buildscript is intended to be used. I'm attempting to use the cradio layout (34k split) with nice_nano_v2. Since shields aren't supported in the local build script, I'm having to resort to using the gh actions script (not terrible, but quite annoying :) ). I tried to make a bash script with `yq` that used shields as well, but I'm no bash wizard, so I'll spare you the pain of the non-working code haha.

urob, 2024-04-08:
This would indeed be nice extension. I hadn't needed it so far to look into it. Using a lightweight processor like `yq` would be my preferred route as well.

urob, 2025-02-19:
Closing this as the old build scripts have been depreciated in favor of my `just build` recipe. The latter works for shields, snippets, etc.

closed 2025-02-19 by urob

krlenell, 2025-06-05:
I'm having trouble building for the corne sold by typeractive. 
In my build.yaml I have:
```
include:
  - board: nice_nano_v2
    shield:
      - corne_left
      - nice_view_adapter
      - nice_view
  - board: nice_nano_v2
    shield:
      - corne_right
      - nice_view_adapter
      - nice_view
```
I'm wondering how to set up the keymap and conf to make this work. I mostly have tried just copying the corneishzen configs, but that hasn't been successful.