urob/zmk-config #57: Combos around <E>-key

mtauz, 2024-04-30:
At first - thank you for sharing these great ideas!

I was wondering why all the combos worked except the <FSLH>combo.
Typing <RM2+RB2> resulted in <e><,>.
I had to add a line 

`CONFIG_ZMK_COMBO_MAX_COMBOS_PER_KEY=6`

to my _board.conf_ file, now it works as expected.

urob, 2024-05-01:
> I was wondering why all the combos worked except the `<FSLH>` combo. Typing `<RM2+RB2>` resulted in `<e><,>`. I had to add a line
> 
> `CONFIG_ZMK_COMBO_MAX_COMBOS_PER_KEY=6`
> 
> to my _board.conf_ file, now it works as expected.

Yes, if you are adding additional combos, you may need to update that setting. FYI, my [build script](https://github.com/urob/zmk-config/tree/main/scripts) automatically updates `CONFIG_ZMK_COMBO_MAX_COMBOS_PER_KEY` as needed.

mtauz, 2024-05-01:
I see, thank you for pointing this out!

closed 2024-05-01 by mtauz