urob/zmk-config #65: Outdated png in readme

stijnveenman, 2024-07-01:
First of all, awesome config and awesome work, it was fun reading up on all the different features. 

The keymap PNG in the `README.md` is slightly outdated.

At least the home row mod shift and control al reversed, this confused me for a bit after configuring my own keyboard. 

I am also interested what tool you used to draw the image, since i made more changes in my own config. I would like to create my own similar image.

urob, 2024-07-03:
Yes, sorry, I haven't gotten to update it since I swapped the HRMs. You find the link to the sources in #33.

urob, 2024-07-21:
Fixed in https://github.com/urob/zmk-config/commit/cdf2be5ad2d7d5b8b5f8eaed38b23fa783a5f523

closed 2024-07-21 by urob