urob/zmk-config #77: Issues with your west manifest simplication

AlexGirardDev, 2024-08-16:
It seems the recent changes you made to simplying the west manifest file causes the following error when trying to build any board

```c
Setting MAX_COMBOS_PER_KEY to 6
Setting MAX_KEYS_PER_COMBO to 3
Building firmware for corneish_zen_v2_left...
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base (cached)).
-- Application: /home/alex/git/urob/zmk/app
-- CMake version: 3.29.2
-- Found Python3: /nix/store/jd9j93smd94xn37fwsqg7xy9d7nffb7v-python3-3.11.6-env/bin/python3.11 (found suitable version "3.11.6", minimum required is "3.8") found components: Interpreter
-- Cache files will be written to: /home/alex/.cache/zephyr
-- Zephyr version: 3.5.0 (/home/alex/git/urob/zmk/modules/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- ZMK Config directory: /home/alex/git/urob/config
-- ZMK Config Kconfig: /home/alex/git/urob/config/corneish_zen.conf
-- Using keymap file: /home/alex/git/urob/config/corneish_zen.keymap
-- Board: corneish_zen_v2_left
-- Found host-tools: zephyr 0.16.8 (/nix/store/qpjdbwhjf7c6xqfxv5zl98gq53adlhq5-zephyr-sdk-0.16.8)
-- Found toolchain: zephyr 0.16.8 (/nix/store/qpjdbwhjf7c6xqfxv5zl98gq53adlhq5-zephyr-sdk-0.16.8)
-- Found BOARD.dts: /home/alex/git/urob/zmk/app/boards/arm/corneish_zen/corneish_zen_v2_left.dts
-- Found devicetree overlay: /home/alex/git/urob/config/corneish_zen.keymap
-- Generated zephyr.dts: /home/alex/git/urob/.build/corneish_zen_v2_left/zephyr/zephyr.dts
-- Generated devicetree_generated.h: /home/alex/git/urob/.build/corneish_zen_v2_left/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: /home/alex/git/urob/.build/corneish_zen_v2_left/zephyr/dts.cmake
Parsing /home/alex/git/urob/zmk/app/Kconfig
/home/alex/git/urob/zmk/modules/zephyr/scripts/kconfig/kconfig.py: /home/alex/git/urob/zmk/zephyr/Kconfig:8: recursive 'source' of 'Kconfig.zephyr' detected. Check that environment variables are set correctly.
Include path:
/home/alex/git/urob/zmk/app/Kconfig:725
Kconfig.zephyr:41
modules/Kconfig:13
/home/alex/git/urob/.build/corneish_zen_v2_left/Kconfig/Kconfig.modules:8
/home/alex/git/urob/zmk/zephyr/Kconfig:8
CMake Error at /home/alex/git/urob/zmk/modules/zephyr/cmake/modules/kconfig.cmake:355 (message):
  command failed with return code: 1
Call Stack (most recent call first):
  /home/alex/git/urob/zmk/modules/zephyr/cmake/modules/zephyr_default.cmake:129 (include)
  /home/alex/git/urob/zmk/modules/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  /home/alex/git/urob/zmk/modules/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:97 (include_boilerplate)
  CMakeLists.txt:8 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /nix/store/4rw9jlln3kc55l8ifd4zai0amz10596w-cmake-3.29.2/bin/cmake -DWEST_PYTHON=/nix/store/jd9j93smd94xn37fwsqg7xy9d7nffb7v-python3-3.11.6-env/bin/python3.11 -B/home/alex/git/urob/.build/corneish_zen_v2_left -GNinja -DZMK_CONFIG=/home/alex/git/urob/config -S/home/alex/git/urob/zmk/app
error: Recipe `_build_single` failed with exit code 1
error: Recipe `build` failed with exit code 1
````

If i rollback your changes to the manifest and then do the following

1. rm -rf .west .buld zmk 
2. just init
3. just build zen
4. it will build successfully

But it seems like with manifest changes you need to specfically clear out the ZMK directory and have init redownload it to cause the issue or fix the issue

IE, i have the working version of the manifest file.

1. I build a board on the commit before your manifest changes and its successfull
2. I then apply your manifest simplication commit
3. I can then still build boards without issues
4. but if i delete zmk dir then do another just init
5. now i get the error when trying to build a board

Alternativly if you just do a fresh clone of your current repo and try and build i think the issue sohuld also occur 

Some additional info:
OS: NixOS 24.11.20240802.d049530 (Vicuna) x86_64
Kernel: 6.10.1-zen1

urob, 2024-08-16:
Sorry, I forgot to push a commit for the `going-modular` ZMK branch that addresses this. It should be resolved by https://github.com/urob/zmk/commit/44f5f67f5cc222a3b8682a6687f5deaf5c948f55. 

If you don't want to re-clone, the following steps should allow you to refresh:
- `rm -rf .west`
- `rm -rf zmk`
- restart a new shell
- `west update`

---

FYI: this issue is cause by `zephy` being in the root folder of `zmk`. Originally, I had prevented this by replicating `zmk/app/west.yml` inside of `config/west.yml`. The new (and simplified) approach is to just patch `zmk` instead. The issue you are experiencing is that the patch didn't get pushed.

AlexGirardDev, 2024-08-16:
no worries, thanks for the fast response and fix.

Thanks for all the work you have done, i have been loving your config and your new nix workflow even more!

closed 2024-08-16 by AlexGirardDev