urob/zmk-config #92: Update actions/upload-artifact to v4 in GitHub Actions workflows

Tomedaro, 2025-01-16:
Hi, I noticed that the GitHub Actions workflows in this repository are using actions/upload-artifact@v3, which has been deprecated as of April 16, 2024 ([Deprecation Notice](https://github.blog/changelog/2024-04-16-deprecation-notice-v3-of-the-artifact-actions/)).

As a result, workflows using this action are failing due to the deprecation. Updating to actions/upload-artifact@v4 would resolve this issue and ensure compatibility with current GitHub Actions functionality.

urob, 2025-01-16:
Can you point me to where you are seeing this? 

Both the docker & the nix workflow call reusable workflows that use `v4`:
- https://github.com/zmkfirmware/zmk/blob/e4f1454e0bc9d86540f738c9de11ff9811290ed0/.github/workflows/build-user-config.yml#L167
- https://github.com/urob/zmk-actions/blob/7602bab62c7459bb97bc5f24e2c9b8e131ae8316/.github/workflows/build-user-config.yml#L162

Maybe you are on an old branch?

Tomedaro, 2025-01-18:
You are right! Thank you, and sorry about the false issue. I used: 
```
on: [push, pull_request, workflow_dispatch]

jobs:
  build:
    uses: urob/zmk/.github/workflows/build-user-config.yml@build-with-submodules
```

urob, 2025-01-18:
Yes, I realized after I first replied that older forks might still be using this workflow. 

I didn't notice myself as I stopped using it after upgrading the helpers to v2 and loading them as a west submodule. 

I now added a more informative error message to the workflow so that anyone else running into this should have an easier time figuring out what to do.

closed 2025-01-18 by urob