urob/zmk-config #96: Same combo shortcuts used by RET and smart_mouse

floating-cat, 2025-01-19:
In this change, it seems there is a mistake in https://github.com/urob/zmk-config/commit/125ae6ecf51c9155fef385800de6f14530ebb4fe#diff-e686312a34f41e65aa4cc3228a939731a8afc444bbeb7adf16c6b89740944ea3 which set both `LT2 LT1` to RETURN and &smart_mouse.
Also, it caused a wrong layer picture:

[image]

urob, 2025-01-19:
Good catch, fixed in [cff4eb0](https://github.com/urob/zmk-config/commit/cff4eb0ca33e60058356ac691375fe423e3cd402). Moving smart-mouse to that combo position was intentional. The `RET` binding was a reminiscent from before I had bound `RET` to `LH0`. Now that `RET` is on a primary thumb position, binding the mouse layer seems like a better use for that combo position.

floating-cat, 2025-01-20:
@urob I think your fixed is not enough. In https://github.com/urob/zmk-config/blob/cff4eb0ca33e60058356ac691375fe423e3cd402/config/base.keymap#L122-L127, the LT1 LT3 is in the ignored key list instead of LT1 LT2 so pressing LT1 LT2 would trigger the smart mouse and exit immediately:

urob, 2025-01-20:
Another good catch, hopefully fixed now.

floating-cat, 2025-01-20:
Yes, it is fixed now so I am closing this issue!

closed 2025-01-20 by floating-cat