urob/zmk-config #97: Scroll down doesn't work in mouse layer

floating-cat, 2025-01-19:
Hi,

I created a ZMK configuration for my Sofle keyboard, almost the same as your Corneish Zen keyboard configuration. However, I noticed that when my keyboard is in the smart-mouse layer, the scroll-down key doesn't work (though the scroll-up key on the left works) and switches back to the Base layer.

I suspect this might be a bug in your configuration, but I haven't figured it out. Would you mind checking if your keyboard configuration in your keyboard also has this issue?

Thanks a lot!

[image]

urob, 2025-01-20:
Hard to diagnose without seeing your config. Two things that come to mind:
 
1. Are you sourcing the key-labels header for the Sofle instead of the 36-keys one?
https://github.com/urob/zmk-config/blob/cff4eb0ca33e60058356ac691375fe423e3cd402/config/corneish_zen.keymap#L11

2. In case you moved the bindings on the mouse layer, you would also need to adjust what keys are included in the `ignored-key-positions` property of `smart-mouse`:
https://github.com/urob/zmk-config/blob/cff4eb0ca33e60058356ac691375fe423e3cd402/config/base.keymap#L125-L126

closed 2025-01-20 by floating-cat

floating-cat, 2025-01-20:
@urob Really thanks! Yes, I hadn't added the needed ignored keys in the `ignored-key-positions`. After adding it, it now works well.