urob/zmk-config #99: Build failure with ZMK_ADAPTIVE_KEY

anarion80, 2025-01-30:
I can't get the build to work locally after recent updates. The issue:

`devicetree error: /Users/anarion/Documents/Repos/zmk-config-corneish-zen/config/base.keymap:130 (column 1): parse error: expected '/' or label reference (&foo)` 

is with ZMK_ADAPTIVE_KEY which does not seem to be expanded in the resulting devicetree:

```c

/ { behaviors { magic_shift: magic_shift { compatible = "zmk,behavior-hold-tap"; #binding-cells = <2>; bindings = <&kp>, <&magic_shift_tap>; flavor = "balanced"; tapping-term-ms = <200>; quick-tap-ms = <175>; }; }; };

/ { behaviors { magic_shift_tap: magic_shift_tap { compatible = "zmk,behavior-mod-morph"; #binding-cells = <0>; bindings = <&shift_repeat>, <&caps_word>; mods = <(0x02)>; }; }; };

ZMK_ADAPTIVE_KEY(
    shift_repeat, bindings = <&sk (((((0x07) << 16) | (0xE1))))>;
    repeat {
      trigger-keys = <((((0x07) << 16) | (0x04))) ((((0x07) << 16) | (0x05))) ((((0x07) << 16) | (0x06))) ((((0x07) << 16) | (0x07))) ((((0x07) << 16) | (0x08))) ((((0x07) << 16) | (0x09))) ((((0x07) << 16) | (0x0A))) ((((0x07) << 16) | (0x0B))) ((((0x07) << 16) | (0x0C))) ((((0x07) << 16) | (0x0D))) ((((0x07) << 16) | (0x0E))) ((((0x07) << 16) | (0x0F))) ((((0x07) << 16) | (0x10))) ((((0x07) << 16) | (0x11))) ((((0x07) << 16) | (0x12))) ((((0x07) << 16) | (0x13))) ((((0x07) << 16) | (0x14))) ((((0x07) << 16) | (0x15))) ((((0x07) << 16) | (0x16))) ((((0x07) << 16) | (0x17))) ((((0x07) << 16) | (0x18))) ((((0x07) << 16) | (0x19))) ((((0x07) << 16) | (0x1A))) ((((0x07) << 16) | (0x1B))) ((((0x07) << 16) | (0x1C))) ((((0x07) << 16) | (0x1D)))>;
      bindings = <&key_repeat>;
      max-prior-idle-ms = <1200>;
      strict-modifiers;
    };)
&caps_word {
  /delete-property/ ignore-modifiers;
};
```

I have cleaned everything, done init, upgrade, update, but no change.

The modules are there locally:
```console
❯ ll modules/zmk/adaptive-key
.rw-r--r--   71 anarion staff 20 Dec  2024 --  .clang-format
drwxr-xr-x    - anarion staff 30 Jan 13:08 -I  .git
drwxr-xr-x    - anarion staff 20 Dec  2024 --  .github
.rw-r--r--  219 anarion staff 20 Dec  2024 --  CMakeLists.txt
drwxr-xr-x    - anarion staff 20 Dec  2024 --  dts
drwxr-xr-x    - anarion staff 20 Dec  2024 --  include
.rw-r--r--  685 anarion staff 20 Dec  2024 --  Kconfig
.rw-r--r-- 1.1k anarion staff 20 Dec  2024 --  LICENSE
.rw-r--r-- 7.9k anarion staff 20 Dec  2024 --  README.md
drwxr-xr-x    - anarion staff 20 Dec  2024 --  src
drwxr-xr-x    - anarion staff 20 Dec  2024 --  tests
drwxr-xr-x    - anarion staff 20 Dec  2024 --  zephyr
```

Logs:
```console
❯ just build all
Setting MAX_COMBOS_PER_KEY to 6
Setting MAX_KEYS_PER_COMBO to 3
Building firmware for corneish_zen_v1_left...
-- west build: generating a build system
Loading Zephyr default modules (Zephyr base).
-- Application: /Users/anarion/Documents/Repos/zmk-config-corneish-zen/zmk/app
-- CMake version: 3.31.3
-- Found Python3: /nix/store/hz1m2axs6cdyh39khp7amj2hq23xscd9-python3-3.12.8-env/bin/python3.12 (found suitable version "3.12.8", minimum required is "3.8") found components: Interpreter
-- Cache files will be written to: /Users/anarion/Library/Caches/zephyr
-- Zephyr version: 3.5.0 (/Users/anarion/Documents/Repos/zmk-config-corneish-zen/zephyr)
-- Found west (found suitable version "1.3.0", minimum required is "0.14.0")
-- ZMK Config directory: /Users/anarion/Documents/Repos/zmk-config-corneish-zen/config
-- ZMK Config Kconfig: /Users/anarion/Documents/Repos/zmk-config-corneish-zen/config/corneish_zen.conf
-- Using keymap file: /Users/anarion/Documents/Repos/zmk-config-corneish-zen/config/corneish_zen.keymap
-- Using keymap file: /Users/anarion/Documents/Repos/zmk-config-corneish-zen/config/corneish_zen.keymap
-- Board: corneish_zen_v1_left
-- Found host-tools: zephyr 0.16.9 (/nix/store/jab7xi6rshbi0c7d9mkw65imr6yy3iga-zephyr-sdk-0.16.9)
-- Found toolchain: zephyr 0.16.9 (/nix/store/jab7xi6rshbi0c7d9mkw65imr6yy3iga-zephyr-sdk-0.16.9)
-- Found Dtc: /nix/store/86kq788n8jp03vlnbhhg983s8hjgazf1-dtc-1.7.2/bin/dtc (found suitable version "1.7.2", minimum required is "1.4.6")
-- Found BOARD.dts: /Users/anarion/Documents/Repos/zmk-config-corneish-zen/zmk/app/boards/arm/corneish_zen/corneish_zen_v1_left.dts
-- Found devicetree overlay: /Users/anarion/Documents/Repos/zmk-config-corneish-zen/config/corneish_zen.keymap
devicetree error: /Users/anarion/Documents/Repos/zmk-config-corneish-zen/config/base.keymap:130 (column 1): parse error: expected '/' or label reference (&foo)
-- In: /Users/anarion/Documents/Repos/zmk-config-corneish-zen/.build/corneish_zen_v1_left/zephyr, command: /nix/store/hz1m2axs6cdyh39khp7amj2hq23xscd9-python3-3.12.8-env/bin/python3.12;/Users/anarion/Documents/Repos/zmk-config-corneish-zen/zephyr/scripts/dts/gen_defines.py;--dts;/Users/anarion/Documents/Repos/zmk-config-corneish-zen/.build/corneish_zen_v1_left/zephyr/zephyr.dts.pre;--dtc-flags;'-Wno-unique_unit_address_if_enabled';--bindings-dirs;/Users/anarion/Documents/Repos/zmk-config-corneish-zen/modules/zmk/adaptive-key/dts/bindings;/Users/anarion/Documents/Repos/zmk-config-corneish-zen/modules/zmk/auto-layer/dts/bindings;/Users/anarion/Documents/Repos/zmk-config-corneish-zen/modules/zmk/leader-key/dts/bindings;/Users/anarion/Documents/Repos/zmk-config-corneish-zen/modules/zmk/tri-state/dts/bindings;/Users/anarion/Documents/Repos/zmk-config-corneish-zen/zmk/app/module/dts/bindings;/Users/anarion/Documents/Repos/zmk-config-corneish-zen/zmk/app/dts/bindings;/Users/anarion/Documents/Repos/zmk-config-corneish-zen/zephyr/dts/bindings;--header-out;/Users/anarion/Documents/Repos/zmk-config-corneish-zen/.build/corneish_zen_v1_left/zephyr/include/generated/devicetree_generated.h.new;--dts-out;/Users/anarion/Documents/Repos/zmk-config-corneish-zen/.build/corneish_zen_v1_left/zephyr/zephyr.dts.new;--edt-pickle-out;/Users/anarion/Documents/Repos/zmk-config-corneish-zen/.build/corneish_zen_v1_left/zephyr/edt.pickle;--vendor-prefixes;/Users/anarion/Documents/Repos/zmk-config-corneish-zen/zmk/app/dts/bindings/vendor-prefixes.txt;--vendor-prefixes;/Users/anarion/Documents/Repos/zmk-config-corneish-zen/zephyr/dts/bindings/vendor-prefixes.txt
CMake Error at /Users/anarion/Documents/Repos/zmk-config-corneish-zen/zephyr/cmake/modules/dts.cmake:279 (message):
  gen_defines.py failed with return code: 1
Call Stack (most recent call first):
  /Users/anarion/Documents/Repos/zmk-config-corneish-zen/zephyr/cmake/modules/zephyr_default.cmake:129 (include)
  /Users/anarion/Documents/Repos/zmk-config-corneish-zen/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  /Users/anarion/Documents/Repos/zmk-config-corneish-zen/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:9 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: /nix/store/52baknjns6l6jfd4fj1imbq9n3qr79mf-cmake-3.31.3/bin/cmake -DWEST_PYTHON=/nix/store/hz1m2axs6cdyh39khp7amj2hq23xscd9-python3-3.12.8-env/bin/python3.12 -B/Users/anarion/Documents/Repos/zmk-config-corneish-zen/.build/corneish_zen_v1_left -GNinja -DBOARD=corneish_zen_v1_left -DZMK_CONFIG=/Users/anarion/Documents/Repos/zmk-config-corneish-zen/config -S/Users/anarion/Documents/Repos/zmk-config-corneish-zen/zmk/app
error: Recipe `_build_single` failed with exit code 1
error: Recipe `build` failed with exit code 1

```

Strangely it works in Github Action: https://github.com/anarion80/zmk-config-corneish-zen/actions/runs/13052011270

My project: https://github.com/anarion80/zmk-config-corneish-zen/tree/pull_upstream

anarion80, 2025-01-31:
Indeed, `helper.h` is old and does not contain adaptive key macro.

`just update` still gets the old version from December:

```console
...
❯ just init
west init -l config
=== Initializing from existing manifest repository config
--- Creating /Users/anarion/Documents/Repos/zmk-config-corneish-zen/.west and local configuration file
=== Initialized. Now run "west update" inside /Users/anarion/Documents/Repos/zmk-config-corneish-zen.
west update --fetch-opt=--filter=blob:none
=== updating zmk (zmk):
--- zmk: initializing
Initialized empty Git repository in /Users/anarion/Documents/Repos/zmk-config-corneish-zen/zmk/.git/
--- zmk: fetching, need revision v0.1+mouse
remote: Enumerating objects: 17664, done.
remote: Total 17664 (delta 0), reused 0 (delta 0), pack-reused 17664 (from 1)
Receiving objects: 100% (17664/17664), 2.34 MiB | 4.95 MiB/s, done.
Resolving deltas: 100% (11945/11945), done.
From https://github.com/urob/zmk
 * tag                 v0.1+mouse -> FETCH_HEAD
 * [new tag]           v0.1       -> v0.1
 * [new tag]           v0.1+mouse -> v0.1+mouse
 * [new tag]           v0.1.0     -> v0.1.0
remote: Enumerating objects: 2087, done.
remote: Counting objects: 100% (890/890), done.
remote: Compressing objects: 100% (570/570), done.
remote: Total 2087 (delta 473), reused 320 (delta 320), pack-reused 1197 (from 1)
Receiving objects: 100% (2087/2087), 4.68 MiB | 9.34 MiB/s, done.
Resolving deltas: 100% (1000/1000), done.
Updating files: 100% (2530/2530), done.
HEAD is now at 84baf929 fix(pointing): Complete header rename missed in refactor (#2702)
HEAD is now at 84baf929 fix(pointing): Complete header rename missed in refactor (#2702)
=== updating zephyr (zephyr):
--- zephyr: fetching, need revision v3.5.0+zmk-fixes
remote: Total 0 (delta 0), reused 0 (delta 0), pack-reused 0 (from 0)
From https://github.com/urob/zephyr
 * branch              v3.5.0+zmk-fixes -> FETCH_HEAD
HEAD is now at 0562599c scripts: check_init_priorities: drop recursive child parsing
=== updating zmk-adaptive-key (modules/zmk/adaptive-key):
HEAD is now at 2b2486b Bump ZMK to v0.1.0 (#3)
=== updating zmk-auto-layer (modules/zmk/auto-layer):
HEAD is now at 5da579c Bump ZMK to v0.1.0 (#3)
=== updating zmk-helpers (modules/zmk/helpers):
HEAD is now at 8db2b01 Bump ZMK to v0.1.0 (#64)
...
```

After deleting `modules/zmk/helpers`, cleaning and running `just init again` all works again. Thanks.

closed 2025-01-31 by anarion80

urob, 2025-01-31:
Hm, that's strange. It sounds like `west update` somehow didn't see that the revision tag (minor version) was moved to a new commit.