urob/zmk-config #100: Layer toggle-modes doesn't work

kotkota, 2025-02-11:
Looks like [#2555](https://github.com/zmkfirmware/zmk/pull/2555) is not merged

urob, 2025-02-11:
The latest upstream release ([v0.1](https://github.com/zmkfirmware/zmk/releases/tag/v0.1.0)) doesn't yet include https://github.com/zmkfirmware/zmk/pull/2555. It'll be part of [v0.2](https://github.com/zmkfirmware/zmk/pull/2693) once it is released. 

In the meantime, you could switch to the current development version of zmk by replacing this block in your west manifest https://github.com/urob/zmk-config/blob/4d00a15b88a4647ff1d09515afac969567bfb306/config/west.yml#L33-L35 with
```yml
    - name: zmk
      revision: main
      remote: zmkfirmware
      import: app/west.yml
```

closed 2025-02-11 by urob

kotkota, 2025-02-12:
Got it. Thank you