urob/zmk-config #109: Compatibility Issue with poetry2nix and Latest Keymap Drawer

hitsmaxft, 2025-04-02:
Hi @urob,

I wanted to report an issue regarding the Keymap Drawer dependency in your ZMK config project. It appears the current poetry2nix implementation doesn't support Poetry 2.0+ versions, which prevents using the latest Keymap Drawer updates.

What I've tried:

    Successfully worked around this by switching from poetry2nix to uv2nix
    Managed to update to the latest Keymap Drawer version using this alternative approach

Question:
Would you consider:

    Officially migrating to a different dependency manager that supports newer Poetry versions?
    Adding alternative installation methods for Keymap Drawer?
    Or have other suggestions for maintaining compatibility?

I'm happy to share more details about my uv2nix implementation if helpful. What are your thoughts on this compatibility challenge?

here is my fork of keymap drawer  https://github.com/hitsmaxft/keymap-drawer

urob, 2025-04-02:
Did you see https://github.com/urob/zmk-config/commit/b460b8f6390e8eedd9c1de1212fa8464cc1c690a? This works pretty well for me and doesn't require forking keymap drawer.

(Note: there is a small typo in that I set the version to `0.20.0` but the rev points to `0.2.0` so that ends up being without consequence (other than `keymap -v` reporting a wrong version).)

hitsmaxft, 2025-04-02:
I've found cause ofthis problem that cause by  use a relase version of nixpkgs , since poetry in nixpkgs release 24.11 sill 1.9.x .

urob, 2025-04-02:
Yes, sorry, I should have mentioned that it requires `nixos-unstable`. I thought I switched to unstable in the linked commit. But I guess had already done so in preparation a few days prior in https://github.com/urob/zmk-config/commit/baaa607e68a5caa8556ab342b3d9954c51731ca4 and then just waited for Hydra to bump `poetry-core`.

Re: `gcovrr`, someone else pointed that out in #108. This must be a recent regression in `nixos-unstable`. If you use the version pinned in `flake.lock` the build should succeed. But it would be good to know what package caused the regression so we can track or open any related upstream issue.

urob, 2025-04-02:
OK, this seems to be an issue with https://github.com/pyproject-nix/pyproject.nix, which doesn't account for the move of `gcovr` in https://github.com/NixOS/nixpkgs/pull/386508.

urob, 2025-04-02:
Fixed upstream, just bumped `nixpkgs` and all tests are green (https://github.com/urob/zmk-config/actions/runs/14228038726)

closed 2025-04-02 by urob