urob/zmk-config #111: ModuleNotFoundError: No module named 'pkg_resources'

te9no, 2025-04-06:
Hi @urob,

I was intrigued by 'Bonus: A (moderately) faster Github Actions Workflow' and tried using it with my own code, but the following error occurred. 
```
[15/604] Generating nanopb/generator/proto/nanopb_pb2.py
FAILED: nanopb/generator/proto/nanopb_pb2.py /home/runner/work/_temp/build.pClLEepF4R/nanopb/generator/proto/nanopb_pb2.py 
cd /home/runner/work/_temp/build.pClLEepF4R && /home/runner/work/_temp/tests/modules/lib/nanopb/generator/protoc -I/home/runner/work/_temp/build.pClLEepF4R/nanopb/generator/proto --python_out=/home/runner/work/_temp/build.pClLEepF4R/nanopb/generator/proto /home/runner/work/_temp/build.pClLEepF4R/nanopb/generator/proto/nanopb.proto
Traceback (most recent call last):
  File "/home/runner/work/_temp/tests/modules/lib/nanopb/generator/protoc", line 13, in <module>
    from proto._utils import invoke_protoc
  File "/home/runner/work/_temp/tests/modules/lib/nanopb/generator/proto/__init__.py", line 10, in <module>
    import pkg_resources
ModuleNotFoundError: No module named 'pkg_resources'
```

The build passes with the setting reset firmware. 
I searched Google for related errors and found the following page: https://stackoverflow.com/questions/7446187/no-module-named-pkg-resources

I cloned zmk-action and tried running `sudo pip uninstall -y setuptools` and `sudo pip install setuptools==69.5.1`, but the error persists. 
https://github.com/te9no/zmk-actions/blob/main/.github/workflows/build-user-config.yml#L76-L80

Since this is my first time creating an issue, I apologize in advance if I've misunderstood or made an error.

urob, 2025-04-06:
Please provide more context:
- how does the entry point workflow look that calls the reusable workflow?
- what board are you trying to build?
- how does your west.yml look?

urob, 2025-04-07:
You'd need to add `setuptools` to the list of python packages in `zephyr-sdk/flake.nix` in the actions repo. It's just a bit of mystery to me why it isn't part of the zephyr requirements file.

te9no, 2025-04-07:
Thank you for your comment. 
As per your suggestion, I added `setuptools` to `setup-sdk/flake.nix`. 
https://github.com/te9no/zmk-actions/blob/main/setup-sdk/flake.nix#L29

Upon building, it seems that setuptools is being added by nix. 
https://github.com/te9no/zmk-config-torq/actions/runs/14300280568/job/40074421174#step:3:197

However, the error remains the same: `ModuleNotFoundError: No module named 'pkg_resources'. `
https://github.com/te9no/zmk-config-torq/actions/runs/14300280568/job/40074421174#step:6:176

Since the build works with the usual action, this issue is not of high priority and might be a personal problem. 
If you think it should be closed, please let me know.

urob, 2025-04-07:
Let's keep it open in case others run into this.

Your edit looks correct and I verified locally that this indeed provides the `pkg_resources` module on `py12`. Why it isn't picked up by `nanopb` is unclear. 

(On a sidenote: newer versions of `nanopb` remove the `pkg_resources` dependency, so eventually this will resolve itself when ZMK bumps zephyr).

te9no, 2025-04-11:
Hello,

I have discovered one thing and would like to share the information with you.
As shown below, I have tried to ensure that only the ZMK repository is referenced. 
https://github.com/te9no/zmk-config-geaconproto/blob/test/config/west.yml

When I built this code, the build was successful even in the Nix environment! 
https://github.com/te9no/zmk-config-geaconproto/actions/runs/14395226985/job/40369623750

Considering the way you describe the modules in your ZMK-config repository's west.yml, I think I might be mistaken in how I reference the modules in the west.yml file.

te9no, 2025-04-11:
The cause has been identified! 
It seems that including CONFIG_ZMK_STUDIO=y in the .conf file leads to the issue error. 
It has been confirmed that builds can be done successfully using the following test repository: 
https://github.com/te9no/zmk-config-roBa/actions/runs/14401365392

Is ZMK Studio not usable with nix builds?

te9no, 2025-04-12:
When using the nix workflow, the build succeeded only when studio was disabled, regardless of whether a module was being referenced. 
Similarly, when building against upstream/main, disabling studio results in a successful build, 
whereas enabling studio causes the build to fail with a `“ModuleNotFoundError: No module named 'pkg_resources'” `error.

That said, the nix build is impressively fast and satisfying to work with. 
Many thanks for developing and sharing it!

urob, 2025-04-13:
Try this:

```yml
name: Build ZMK firmware (nix)
on:
  workflow_dispatch:
  push:
    paths:
      - config/**
      - build.yaml

jobs:
  build:
    uses: urob/zmk-actions/.github/workflows/build-user-config.yml@main
    with:
      toolchain: zephyr-full
```

te9no, 2025-04-13:
Thank you for confirming. 
I tried it out, but it seems to stop due to another error now.
`error: flake 'path:/home/runner/work/_actions/urob/zmk-actions/v4/setup-sdk' does not provide attribute 'devShells.x86_64-linux.zephyr-full', 'packages.x86_64-linux.zephyr-full', 'legacyPackages.x86_64-linux.zephyr-full' or 'zephyr-full'`
https://github.com/te9no/zmk-config-geaconproto/actions/runs/14425465469/job/40453712464#step:3:105

urob, 2025-04-13:
Make sure you also update the `uses` line as above

urob, 2025-04-13:
Sorry, my bad, I have v4 hardcoded in the workflow on main

urob, 2025-04-13:
Just pushed another commit which hopefully fixes it. Otherwise I'll need to check tomorrow (on mobile right now)

te9no, 2025-04-13:
Wonderful! 
Enabling ZMK Studio allowed the build to succeed. 
Thank you so much. 
https://github.com/te9no/zmk-config-geaconproto/actions/runs/14425888322

closed 2025-04-13 by te9no